pwmckenna/react-native-motion-manager #3: Blank screen

grischoun, 2015-06-27:
Hi,

I am trying to use your project in my own project but I have the following issue.
The application starts fine and I get the following output on the Xcode console.

2015-06-27 12:37:16.556 SlideTracker[2157:12417] Accelerometer
2015-06-27 12:37:16.558 SlideTracker[2157:12417] Accelerometer not Available!
2015-06-27 12:37:16.559 SlideTracker[2157:12417] Magnetometer
2015-06-27 12:37:16.561 SlideTracker[2157:12417] Magnetometer not Available!
2015-06-27 12:37:16.562 SlideTracker[2157:12417] Gyroscope
2015-06-27 12:37:16.563 SlideTracker[2157:12417] Gyroscope not Available!

But on the IPhone the screen remains blank. It never shows the main page of my application. (The behavior is the same when I run your project on its own.)

Any idea on what is wrong?

pwmckenna, 2015-07-25:
Sadly none of these are available in the emulator, so you can only test on actual devices. I don't have any hunches about the blank screen though. Is there any logging/output that you're seeing?

grischoun, 2015-08-08:
Thanks for your reply.

I had tried on the device before but without Chrome debugging on and got the blank screen.
I have just tried again and with debugging on and I got the following error:

Uncaught TypeError: Cannot read property 'apply' of undefinedmessageHandlers.executeJSCall @ debugger-ui:54ws.onmessage @ debugger-ui:80

This explains why it is stuck.

By googling around I think the issue is linked to versioning but I am not sure. My project uses react-native@0.5.0 and react-native-motion-manager needs @0.3.11, But when I change my project's react-native version to 0.3.11 there is a linking problem in Xcode.

grischoun, 2015-08-08:
I think I fixed it by removing the dependency on react-native in your project.
No more blank screen, neither on the simulator nor the device.

closed 2015-08-08 by grischoun