pwmckenna/react-native-motion-manager #7: Expose access to attitude data

paramaggarwal, 2015-08-15:
From [CMDeviceMotion docs](https://developer.apple.com/library/ios/documentation/CoreMotion/Reference/CMDeviceMotion_Class/index.html):

> An instance of CMDeviceMotion encapsulates measurements of the attitude, rotation rate, and acceleration of a device.

Currently can only get Accelerometer, Gyroscope and Magnometer separately. Attitude data gives combined data about motion, which is super useful to say make an virtual reality app.

(Raised this to keep track of it, not asking for a fix - I do understand that it is the community's responsibility to help out.)

paramaggarwal, 2015-08-27:
I made https://github.com/paramaggarwal/react-native-device-motion for this.

closed 2015-08-27 by paramaggarwal

pwmckenna, 2015-10-14:
that looks awesome!