pwmckenna/react-native-motion-manager #9: Error: Naming collision detected

williambout, 2015-12-13:
```Error building DependencyGraph:
 Error: Naming collision detected: node_modules/react-native-motion-manager/node_modules/react-native/packager/react-packager/src/DependencyResolver/haste/polyfills/polyfills.js collides with node_modules/react-native/packager/react-packager/src/Resolver/polyfills/polyfills.js```

bpolonia, 2016-01-05:
@williambout Have you been able to get this package to work?

@pwmckenna Can you help us with this?

williambout, 2016-01-05:
@bpolonia Yes, just delete the `react-native` folder in `node_modules/react-native-motion-manager/`

pwmckenna, 2016-01-07:
Its entirely possible that the react peer dependency is out of date. Are you using npm 2? If so, it's probably installing your project's react-native dependency which is probably more recent than the peer dep defined here, so npm 2 goes ahead and installs another older copy for this module.

I'd be curious if this repros using npm 3. Regardless the peer dep for this package should be updated to include the newer versions of react-native once its been tested with them. I could use some help on that front.

bpolonia, 2016-01-07:
Hi William. That worked. Thanks!

pwmckenna, 2016-01-08:
just published v0.0.6 which just allowed any version of the react-native dep. I'm going to hope for the best and close this for now, but let me know if anyone runs into trouble with this again.

closed 2016-01-08 by pwmckenna

williambout, 2016-01-08:
Problem solved thanks @pwmckenna 👍