pwmckenna/react-native-motion-manager #11: Gyroscope

bpolonia, 2016-01-11:
I'm trying to call a function only when the gyroscope X axis goes beyond X. As I change the angle of the phone, X changes accordingly but then resets. 

For example I would like to be able to hold the phone at a right angle, and when I lay the phone flat on a table call a function.

But the way your package is working now is I hold the phone up right, the X axis marks 0, then as I lay the phone down X changes accordingly but immediately resets to 0 as the phone stops moving. This is causing a continuous call of my function. 

Any input you can provide?

r00ger, 2016-01-13:
I see the same behaviour

pwmckenna, 2016-01-14:
Sounds like an interesting use case. I'd prefer this package remain a super thin wrapper around the native libraries, but it seems like you could build off of this project to achieve what you're looking for. I don't have any good examples of react-native libraries building on other react-native libraries though...perhaps some research to be done there. I'm going to close this for now, but I'd love to see what you come up with!

pwmckenna, 2016-01-14:
btw, if this turns out to not match the behavior of the underlying native functionality, please re-open this issue and I'll take a look at it.

closed 2016-01-14 by pwmckenna