pwmckenna/react-native-motion-manager #15: Very chatty logs

corbt, 2016-04-19:
Thanks for this library, it's allowing me to implement accelerometer-based functionality easily.

I've noticed that the logs are extremely chatty. I'm getting lines like the following on every sensor update:

```
startAccelerometerUpdates: -0.029541, -0.031815, -1.006012
```

This makes it impossible to use the logs for anything else, because anything besides these updates are quickly pushed off my screen.

Would you mind changing the default logging behavior of this library to not output any updates at all, unless something breaks?

pwmckenna, 2016-04-23:
@corbt don't have the time to do it myself, but would happily accept a pr. cheers

Kouznetsov, 2018-03-20:
Incoming

Kouznetsov, 2018-03-20:
Just submitted a PR :)