pwmckenna/react-native-motion-manager #24: Example code needs update

Efreeto, 2016-09-30:
The example code is already two years old and does not compile with the newer react native structure.

Btw, I found 'react-native-location' module a very useful alternative to this library.

pwmckenna, 2016-10-10:
Hi @Efreeto, I would welcome any updates to the examples. I also don't use this library anymore, so I don't do much maintenance, and would happily recommend another library if it would be helpful to folks. Feel free to make a PR with that recommendation as well :)

Efreeto, 2016-10-10:
Thanks for understanding @pwmckenna. Your library still works well, but it's just the example code that no longer compiles. Also your library is the only react native library for getting the raw values as far as I know. The other library is useful for getting a bit higher level values.

closed 2016-10-10 by Efreeto