pwmckenna/react-native-motion-manager #25: Cannot read property 'setGyroUpdateInterval of undefined'

Bobby-hewitt, 2016-11-01:
I keep getting this error and occasionally a message saying 'undefined is not an object'.  I have installed it as instructed but can't seem to get around this issue.  Any help would be appreciated :)

peterept, 2017-03-23:
This is fixed in pending [PR#31](https://github.com/pwmckenna/react-native-motion-manager/pull/31).

peterept, 2017-03-23:
To use that forked PR directly instead of this repo use:

`$ npm install git://github.com/isaacnass/react-native-motion-manager.git --save`
`$ react-native link`

Then you are all set.

armenr, 2017-04-01:
@peterept -- I'm getting the following with both this repo, and your fork of this project (BTW, I ended up doing the same work you did of manually fixing/updating the declarations to work with the latest XCode, but thank you for making your work visible for the rest of us!)

[image]
[image]