pwmckenna/react-native-motion-manager #29: errors when building project

gwmccull, 2017-01-06:
Hi, I'm getting a bunch of build errors when I try to use react-native-motion-manager.

I created a brand new project with `react-native init` using v0.40.0 of RN.  It builds cleanly (other than some warnings).

Then I install react-native-motion-manager per the instructions and the build fails.

I get the same set of errors for gyroscope, accelerometer and magnetometer.  Here are the errors for gyroscope:
```
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:8:
In file included from ../react-native/React/Base/RCTEventDispatcher.h:12:
/Users/gmccullough/Development/Projects/accelerometer/ios/build/Build/Products/Debug-iphonesimulator/include/React/RCTBridge.h:65:1: error: duplicate interface definition for class 'RCTBridge'
@interface RCTBridge : NSObject <RCTInvalidating>
^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:7:
../react-native/React/Base/RCTBridge.h:65:12: note: previous definition is here
@interface RCTBridge : NSObject <RCTInvalidating>
           ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:8:
In file included from ../react-native/React/Base/RCTEventDispatcher.h:12:
/Users/gmccullough/Development/Projects/accelerometer/ios/build/Build/Products/Debug-iphonesimulator/include/React/RCTBridge.h:154:55: error: property has a previous declaration
@property (nonatomic, copy, readonly) NSArray<Class> *moduleClasses;
                                                      ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:7:
../react-native/React/Base/RCTBridge.h:154:55: note: property declared here
@property (nonatomic, copy, readonly) NSArray<Class> *moduleClasses;
                                                      ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:8:
In file included from ../react-native/React/Base/RCTEventDispatcher.h:12:
/Users/gmccullough/Development/Projects/accelerometer/ios/build/Build/Products/Debug-iphonesimulator/include/React/RCTBridge.h:159:48: error: property has a previous declaration
@property (nonatomic, strong, readonly) NSURL *bundleURL;
                                               ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:7:
../react-native/React/Base/RCTBridge.h:159:48: note: property declared here
@property (nonatomic, strong, readonly) NSURL *bundleURL;
                                               ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:8:
In file included from ../react-native/React/Base/RCTEventDispatcher.h:12:
/Users/gmccullough/Development/Projects/accelerometer/ios/build/Build/Products/Debug-iphonesimulator/include/React/RCTBridge.h:165:37: error: property has a previous declaration
@property (nonatomic, strong) Class executorClass;
                                    ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-nativ
e-motion-manager/RNMotionManager/Gyroscope.m:7:
../react-native/React/Base/RCTBridge.h:165:37: note: property declared here
@property (nonatomic, strong) Class executorClass;
                                    ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:8:
In file included from ../react-native/React/Base/RCTEventDispatcher.h:12:
/Users/gmccullough/Development/Projects/accelerometer/ios/build/Build/Products/Debug-iphonesimulator/include/React/RCTBridge.h:170:61: error: property has a previous declaration
@property (nonatomic, weak, readonly) id<RCTBridgeDelegate> delegate;
                                                            ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:7:
../react-native/React/Base/RCTBridge.h:170:61: note: property declared here
@property (nonatomic, weak, readonly) id<RCTBridgeDelegate> delegate;
                                                            ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:8:
In file included from ../react-native/React/Base/RCTEventDispatcher.h:12:
/Users/gmccullough/Development/Projects/accelerometer/ios/build/Build/Products/Debug-iphonesimulator/include/React/RCTBridge.h:175:53: error: property has a previous declaration
@property (nonatomic, copy, readonly) NSDictionary *launchOptions;
                                                    ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:7:
../react-native/React/Base/RCTBridge.h:175:53: note: property declared here
@property (nonatomic, copy, readonly) NSDictionary *launchOptions;
                                                    ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:8:
In file included from ../react-native/React/Base/RCTEventDispatcher.h:12:
/Users/gmccullough/Development/Projects/accelerometer/ios/build/Build/Products/Debug-iphonesimulator/include/React/RCTBridge.h:180:56: error: property has a previous declaration
@property (nonatomic, readonly, getter=isLoading) BOOL loading;
                                                       ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:7:
../react-native/React/Base/RCTBridge.h:180:56: note: property declared here
[... 16 lines cut ...]
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:7:
../react-native/React/Base/RCTBridge.h:190:63: note: property declared here
@property (nonatomic, readonly, strong) RCTPerformanceLogger *performanceLogger;
                                                              ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:9:
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.h:7:
../react-native/React/Base/RCTBridgeModule.h:55:11: warning: duplicate protocol definition of 'RCTBridgeModule' is ignored
@protocol RCTBridgeModule <NSObject>
          ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:7:
In file included from ../react-native/React/Base/RCTBridge.h:13:
/Users/gmccullough/Development/Projects/accelerometer/ios/build/Build/Products/Debug-iphonesimulator/include/React/RCTBridgeModule.h:55:11: note: previous definition is here
@protocol RCTBridgeModule <NSObject>
          ^
/Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:85:39: warning: 'sendDeviceEventWithName:body:' is deprecated: Subclass RCTEventEmitter instead [-Wdeprecated-declarations]
         [self.bridge.eventDispatcher sendDeviceEventWithName:@"GyroData" body:@{
                                      ^
In file included from /Users/gmccullough/Development/Projects/accelerometer/node_modules/react-native-motion-manager/RNMotionManager/Gyroscope.m:8:
../react-native/React/Base/RCTEventDispatcher.h:83:1: note: 'sendDeviceEventWithName:body:' has been explicitly marked deprecated here
- (void)sendDeviceEventWithName:(NSString *)name body:(id)body
^
2 warnings and 9 errors generated.
```

Any help would be appreciated.

isaacnass, 2017-01-30:
I got you!

`Find and Replace in Project` and change all `#import "RCTBridge.h"` to `#import <React/RCTBridge.h>`

Credit to @SMJ93 in [this issue](https://github.com/facebook/react-native/issues/11725)

gwmccull, 2017-02-11:
Thanks @isaacnass! That worked for me

closed 2017-02-11 by gwmccull